rerun-io/rerun #3156: Large app.ron causes significant slowdown during shutdown

jleibs, 2023-08-30:
**Describe the bug**
Rerun takes 2-3 seconds to shutdown at quit.

**To Reproduce**
Not clear how I got into this state, but the attached app.ron creates the issue for me. This app.ron is currently about 7.2M
[app.ron.zip](https://github.com/rerun-io/rerun/files/12475575/app.ron.zip)

**Expected behavior**
app.ron should stay a reasonable size and rerun should shutdown quickly.

**Screenshots**
[image]

**Backtrace**

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.10

**Rerun version**
`main`

emilk, 2023-08-30:
I'll add some GC:ing to the next egui release

teh-cmc, 2023-08-30:
Tangent question: where are these `3.8s` going? It shouldn't take anywhere near that to write 7M of data to page cache? Is it the RON serializer?

emilk, 2023-09-04:
Will be fixed by:
* https://github.com/emilk/egui/pull/3299

closed 2023-09-04 by emilk