rerun-io/rerun #3157: OPF example: big perf regression between `0.8.1` and tip

teh-cmc, 2023-08-30:
`main`:
[image]

`0.8.1`:
[image]

jleibs, 2023-08-30:
Two more datapoints:

### Intel + Wayland
[image]
Something different (but also not good) going on here.

### Nvidia + XWayland
[image]

teh-cmc, 2023-08-30:
The regression was introduced in https://github.com/rerun-io/rerun/commit/effb72ba15e9faaafcab2c4874e2b287f1e01d5f.

Just bisected into it, haven't looked at the underlying reason yet.

teh-cmc, 2023-08-31:
I cannot reproduce the issue on my laptop: both the embedded intel GPU and the discrete nvidia card give about 55ms per frame.

My laptop has the exact same OS/software setup as my desktop so... I guess this is an AMD thing :unamused:

emilk, 2023-08-31:
The regression diff should only touch CPU code; the GPU should see the exact same data and calls o.O - some weird timing thing?

emilk, 2023-10-09:
For reference, on my MacBook Pro the performance has **improved** by a factor of **3x** from 0.8.2 to 0.9: https://github.com/rerun-io/rerun/issues/1136#issuecomment-1748716579

teh-cmc, 2024-07-09:
Mesa :man_shrugging:

closed 2024-07-09 by teh-cmc