rerun-io/rerun #3202: `LineStripsND` C++ constructors are unfriendly with multiple strips

teh-cmc, 2023-09-04:
I cannot seem to do this with our cpp API:
```cpp
std::vector<std::vector<rr::datatypes::Vec3D>> points = {
    {{0.f, 0.f, 0.f}, {0.f, 0.f, 1.f}},
    {{1.f, 0.f, 0.f}, {1.f, 0.f, 1.f}},
    {{1.f, 1.f, 0.f}, {1.f, 1.f, 1.f}},
    {{0.f, 1.f, 0.f}, {0.f, 1.f, 1.f}},
};
rec.log("segments", rr::LineStrips2D(points));
```

Instead I have to do this, which is quite annoying:
```cpp
std::vector<rr::datatypes::Vec3D> points1 = {{0.f, 0.f, 0.f}, {0.f, 0.f, 1.f}};
std::vector<rr::datatypes::Vec3D> points2 = {{1.f, 0.f, 0.f}, {1.f, 0.f, 1.f}};
std::vector<rr::datatypes::Vec3D> points3 = {{1.f, 1.f, 0.f}, {1.f, 1.f, 1.f}};
std::vector<rr::datatypes::Vec3D> points4 = {{0.f, 1.f, 0.f}, {0.f, 1.f, 1.f}};
rec.log("segments", rr::LineStrips3D({points1, points2, points3, points4}));
```

Not sure if I'm missing some C++ idiom or otherwise.

emilk, 2023-10-09:
This is very related to #3330 

I think we should allow passing any container (something we can call `std::begin/end` on), including `std::vector` and initializer lists.

Wumpf, 2023-10-24:
* Fixed by #3971

closed 2023-10-24 by Wumpf