rerun-io/rerun #3212: Revamp of our example building infrastructure

abey79, 2023-09-05:
The in-app example page (#3191) relies on pre-build RRD for its example. These RRD must be up-to-date for the dev builds, and version-specific for the public release. These RRD are also costly to build, which means we currently have only 7 examples covered. This figure must be increased.

We had a short discussion with @jprochazk and have the following proposal:

- We create a dedicated CI workflow to build RRD examples, based on the existing `build_demo_app.py` script.
- This workflow will be heavy, so it'll be run on a nightly basis of `origin/main`.
- This script will pull _everything_ it needs from example's front matter, including:
  - whether or not a given example must be included in the build
  - any build args to use (e.g. specify dataset, etc.)
  - metadata (title, description, tags, thumbnail, etc.
- Like now, the built examples will be uploaded to our Google storage
  - for each build, under a `commit/XXX` URL path
  - for each build, a symlink/redirect will be updated for `release/nightly`
  - for each release, under `release/0.X.Y`

For the manifest that's baked in the Rerun executable (which contains links to RRD, thumbnails, metadata, etc.), the one under `release/nightly` will be used for dev builds. For release build, a $VERSION variable will be provided and the manifest under `release/$VERSION` will be used instead.

Related:
-  #3190

teh-cmc, 2023-09-05:
Relevant: https://github.com/rerun-io/rerun/pull/3204 introduces the `_RERUN_TEST_FORCE_SAVE=<filepath.rrd>` environment variable, that will force any `RecordingStream` to write to a specified file on disk, irrelevant of how it was set up (`connect()`, `spawn()`, etc...).

emilk, 2023-09-05:
Let's try to get this in before 0.9, but start with just the minimal set of examples.

emilk, 2023-09-11:
The viewer `include_str!`:s `crates/re_viewer/data/examples_manifest.json` which in turn points to `.rrd` urls _at a specific commit_. Currently we need to update `examples_manifest.json` manually.

For 0.9 we could just update it manually to point to .rrd's we've updated specifically for the 0.9 release.

…or we figure out a proper fix for this.

closed 2023-12-14 by jprochazk