rerun-io/rerun #3213: [REMINDER-0.9] Make sure baked-in example manifest is correct

abey79, 2023-09-05:
**This is a reminder issue to ensure that we don't ship 0.9 with the wrong example manifest, as the related infrastructure is still being put in place.**

#3191 introduced an in-app example page that uses a baked-in `example_manifest.json` containing, in particular, links to RRD. We must make sure that these links are "correct" (ie. pointing to RRD built along with the release) for the release executables.

This should be addressed by #3212 and #3190 down the line, but it's likely that this will require manual work for 0.9. This this reminder issue.

emilk, 2023-09-24:
I think this is fixed by https://github.com/rerun-io/rerun/pull/3332, right @jprochazk ?

emilk, 2023-09-25:
* Closed by https://github.com/rerun-io/rerun/pull/3332

closed 2023-09-25 by emilk