rerun-io/rerun #3224: Pre-release check that our `code-example` files don't contain TODO entries.

jleibs, 2023-09-05:
Many of our `code-example`s currently have TODOs in them. While these are helpful to track at the moment, having TODOs show up in the code-examples on the website is a poor user-experience. This isn't something that needs to run on every PR, but it's something we should always check before deploying updated docs with code-examples to the website.

emilk, 2023-09-25:
High-priority to do or remove the TODOs, lower priority to have it as a CI check

emilk, 2023-09-25:
Most of the TODO:s are fixed by:
* #3447

What remains (ignoring C++) is fixed by:
* https://github.com/rerun-io/rerun/issues/3450

emilk, 2023-09-28:
I'm leaving the issue open so we remember to check again before the actual release

Wumpf, 2023-10-04:
The only TODOs right now are in three cpp files about bad ergonomics (as well as roundtrips.py).
Deemed acceptable and highly unlikely that we add anything by accident right now.

closed 2023-10-04 by Wumpf