rerun-io/rerun #3225: Python context/scope object for temporarily setting timestamps

jleibs, 2023-09-05:
Code such as the following recommendation for pre-clearing entities:
```
# Associate the following data with `start_time` on the `time` timeline
rr.set_time_seconds("time", start_time)
# Log the data as usual
rr.log("short_lived", rr.Tensor(one_second_tensor))
# Associate the following clear with `start_time + 1.0` on the `time` timeline
rr.set_time_seconds("time", start_time + 1.0)
rr.log("short_lived", rr.ClearEntity())
```

Can be somewhat risky in the code-base because other data that should still be logged at `start_time` could be logged incorrectly as a result.

It would be much nicer to write this as something like the following:
```
# Associate the following data with `start_time` on the `time` timeline
rr.set_time_seconds("time", start_time)
# Log the data as usual
rr.log("short_lived", rr.Tensor(one_second_tensor))
with rr.Scope():
  # Associate the following clear with `start_time + 1.0` on the `time` timeline
  rr.set_time_seconds("time", start_time + 1.0)
  rr.log("short_lived", rr.ClearEntity())
```
in order to avoid polluting the global timestamp state.

teh-cmc, 2023-09-06:
Related: we already have [support for `with rec:`](https://github.com/rerun-io/rerun/blob/main/rerun_py/rerun_sdk/rerun/recording_stream.py#L22-L27), which is close but not quite the same time, but there's probably an opportunity to make the two cooperate nicely.

abey79, 2025-03-05:
This is better addressed by:
- https://github.com/rerun-io/rerun/issues/3841

closed 2025-03-05 by abey79