rerun-io/rerun #3231: Archetypes should be first-class citizen of the UI

teh-cmc, 2023-09-06:
Currently, the notion of archetype is completely absent from the UI, making it very difficult to map what you see in the viewer back to the logging code and vice versa.

Needs design work.

jleibs, 2023-09-06:
One thing that would potentially simplify some things is to track the usage of an Archetype all the way through rendering at part of the hash that we use for picking.

This would allow us to dispatch hover and selection UIs directly to the handler for the correct archetype instead of getting an entity path and having to try to re-infer the archetype based on its components. In particular, this helps us in cases where a given entity has multiple archetypes. Each of those archetypes might produce a different graphical element but they will both currently have the same InstanceHash.

Additionally, in the future when we introduce the abillity to override archetypes (i.e. force-view this Image as a DepthImage), this should allow us to track how the Entity is being Presented, as distinct from how the entity was logged.

emilk, 2023-09-12:
We now know archetypes based on the indicator components, so showing custom ui:s should be fairly straight-forward to implement.

One important use-case for this is boxes: a user can log a box as XYWH but it is always stored as center + half-extents. Without a specific archetype-ui, the user can never see the data as they logged it.

Wumpf, 2023-09-19:
Box ui can't be solved with that because we can't backtrack right now how a box was logged by looking at its indicator.

Overall the main thing to do here seems to be better handling of indicator components. Otherwise it's a bit unclear how potentially overlapping archetypes should be displayed

Wumpf, 2023-09-22:
For box (and maybe others) we likely want to introduce user configurable view.
Note that this can get fairly complex when it comes to "merging" component batches: Showing an entity with a single box in different representations is relatively straight forward:
[image]

But once `HalfSizes2D` and `Position2D` are both batches, we need a way to select both at once and show them differently. This is not something we have explored previously in the selection ui.

Wumpf, 2025-01-30:
@gavrelina and I discussed this recently and concluded that since archetypes & visualizers should map directly (see https://github.com/rerun-io/rerun/issues/8368) we should put the visualizer section on top and call it something like "Archetype Visualizers"

Furthermore, with the arrival of tags we can show archetype information in the data section (group them by archetype and show field names)

Wumpf, 2025-06-17:
This got solved recently!

closed 2025-06-17 by Wumpf