rerun-io/rerun #3235: Allow for some leeway when comparing floats in `rerun compare`

teh-cmc, 2023-09-06:
E.g. expose a CLI flag or something.

This makes API examples much more natural at it avoids doing crazy stuff just to make the roundtrip tests pass across different languages.

emilk, 2025-03-27:
(copied from a [duplicate issue](https://github.com/rerun-io/rerun/issues/9394)):

A lot of our snippet comparison tests are disabled due to small floating point differences between Python, Rust and C++. We should add an `--aproximate` mode to the `rerun rrd compare` function that ignores small differences.

It doesn't need to be perfect or robust, just good enough for our testing purposes. Something like:

* If the datatype is f32 or f64
* Subtract and see that the abs difference is less than 0.01

emilk, 2025-04-15:
Fixing this would also improve our backwards compatibility tests, by comparing migrated .rrd snippets with the newly produced rrd snippets (which should be the same, within rounding error)

commit 5ced5940 (2025-04-17): Use approximate equals when comparing floats in snippets (#9750)

### Related
* Closes #3235
* Part of https://github.com/rerun-io/rerun/issues/9110
* Unblocks https://github.com/rerun-io/rerun/pull/9751
* Unblocks https://github.com/rerun-io/rerun/issues/9588

### What
In our snippet comparisons, ignore small numeric differences (that often
occur because of differences in programming language, or
compiler/interpreter version)

---------
PR #9750 (2025-04-17): Use approximate equals when comparing floats in snippets
### Related
* Closes #3235
* Part of https://github.com/rerun-io/rerun/issues/9110
* Unblocks https://github.com/rerun-io/rerun/pull/9751
* Unblocks https://github.com/rerun-io/rerun/issues/9588

### What
In our snippet comparisons, ignore small numeric differences (that often occur because of differences in programming language, or compiler/interpreter version)
commit bf9a1c21 (2025-04-17): Use approximate equals when comparing floats in snippets (#9750)

### Related
* Closes #3235
* Part of https://github.com/rerun-io/rerun/issues/9110
* Unblocks https://github.com/rerun-io/rerun/pull/9751
* Unblocks https://github.com/rerun-io/rerun/issues/9588

### What
In our snippet comparisons, ignore small numeric differences (that often
occur because of differences in programming language, or
compiler/interpreter version)

---------
closed 2025-04-17 by emilk