rerun-io/rerun #3238: Panic when opening a RRD

abey79, 2023-09-06:
As of now on `main`, `rerun data.rrd` panics with this RRD: https://demo.rerun.io/commit/5be001c/examples/structure_from_motion/data.rrd

That 5be001c commit was one of my #3191 branch (which is pure UI code), just after merging from `main` yesterday.

Here is the log:
```
[2023-09-06T16:02:08Z INFO  re_data_source::load_file_path] Loading "/Users/hhip/Downloads/data.rrd"…
[2023-09-06T16:02:09Z WARN  re_data_store::store_db] The incoming component "rerun.tensor" had the type:
    Struct(
        [
            Field {
                name: "tensor_id",
                data_type: FixedSizeBinary(
                    16,
                ),
                is_nullable: false,
                metadata: {},
            },
            Field {
                name: "shape",
                data_type: List(
                    Field {
                        name: "item",
                        data_type: Struct(
                            [
                                Field {
                                    name: "size",
                                    data_type: UInt64,
                                    is_nullable: false,
                                    metadata: {},
                                },
                                Field {
                                    name: "name",
                                    data_type: Utf8,
                                    is_nullable: true,
                                    metadata: {},
                                },
                            ],
                        ),
                        is_nullable: false,
                        metadata: {},
                    },
                ),
                is_nullable: false,
                metadata: {},
            },
            Field {
                name: "data",
                data_type: Union(
                    [
                        Field {
                            name: "U8",
                            data_type: Binary,
                            is_nullable: false,
                            metadata: {},
                        },
                        Field {
                            name: "U16",
                            data_type: List(
                                Field {
                                    name: "item",
                                    data_type: UInt16,
                                    is_nullable: false,
                                    metadata: {},
                                },
                            ),
                            is_nullable: false,
                            metadata: {},
                        },
                        Field {
[... 482 lines cut ...]
             at winit-0.28.6/src/platform_impl/macos/event_loop.rs:220:22
      objc2::rc::autorelease::autoreleasepool
             at objc2-0.3.0-beta.3.patch-leaks.3/src/rc/autorelease.rs:313:5
      winit::platform_impl::platform::event_loop::EventLoop<T>::run_return
             at winit-0.28.6/src/platform_impl/macos/event_loop.rs:211:25
  40: <winit::event_loop::EventLoop<T> as winit::platform::run_return::EventLoopExtRunReturn>::run_return
             at winit-0.28.6/src/platform/run_return.rs:51:9
      eframe::native::run::run_and_return
             at eframe/src/native/run.rs:139:16
      eframe::native::run::wgpu_integration::run_wgpu::{{closure}}
             at eframe/src/native/run.rs:1517:17
      eframe::native::run::with_event_loop::{{closure}}
             at eframe/src/native/run.rs:122:9
      std::thread::local::LocalKey<T>::try_with
             at std/src/thread/local.rs:270:16
      std::thread::local::LocalKey<T>::with
             at std/src/thread/local.rs:246:9
      eframe::native::run::with_event_loop
             at eframe/src/native/run.rs:115:16
  41: eframe::native::run::wgpu_integration::run_wgpu
             at eframe/src/native/run.rs:1514:13

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues
```

emilk, 2023-09-07:
The problem will disappear together with `arrow2_convert` when we release 0.9

emilk, 2023-09-25:
This no longer crashes on `main`

closed 2023-09-25 by emilk