rerun-io/rerun #3242: Colormap selection preview is sometimes bugged, sometimes not there at all

teh-cmc, 2023-09-07:
Here's a colormap selector for depth clouds -- no preview at all:
[image]

And here's a colormap selector for tensors -- layout is broken:
[image]

emilk, 2023-09-14:
So there are two problems here:

A) we have two different code paths for selecting a colormap, and only one has the preview
B) the preview is broken

The broken preview is not because of egui, but because how we render it. I'm still investigating.

emilk, 2023-09-14:
The `viewport` given to `new_renderer_callback` for each gradient image is correct:

[image]

So the problem is either in
* how `egui_wgpu` handles the callback (maybe broke in https://github.com/emilk/egui/pull/3253)
* our `rectangle` shader (`re_renderer::renderer::TexturedRect`)
* our transforms (`re_renderer::view_builder::TargetConfiguration`)

PR #3373 (2023-09-21): Fix colormap picker preview & show it for depth point clouds as well
### What

* Fixes #3242

[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3373) (if applicable)

- [PR Build Summary](https://build.rerun.io/pr/3373)
- [Docs preview](https://rerun.io/preview/73ad32ebf606b6d91192ca5c490d0b158c79f769/docs) [... cut]
commit 4a62dd5d (2023-09-21): Fix colormap picker preview & show it for depth point clouds as well (#3373)

### What

* Fixes #3242

[image]

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3373) (if
applicable)

- [PR Build Summary](https://build.rerun.io/pr/3373)
- [Docs
preview](https://rerun.io/preview/73ad32ebf606b6d91192ca5c490d0b158c79f [... cut]
closed 2023-09-21 by emilk