rerun-io/rerun #3260: Complete custom user data story - Tracking issue

teh-cmc, 2023-09-08:
Tracking issue for the complete "custom user data" story as a whole, i.e.:
- The API uses an `as_components()` kind of model:
  - It is used by builtin archetypes and components to drive their serialization.
  - It can be easily implemented in user code, allowing for the creation of new archetypes & components without the IDL.
- Indicator components are materialized in the type system, and can be instantiated in user code with arbitrary names.
- The API provides a way to extend/wrap/embed builtin archetypes and components from user code.
  - E.g. to add a couple custom components to the existing builtin point cloud archetype.
- We provide a `custom_data` example that shows the story as whole.

In general, we should strive to avoid the old "user code" vs. "internal code" dichotomy: user code should be able to do everything that internal code does without having to jump through hoops and without any kind of performance penalties, i.e. there is no such thing as internal code.
This mostly naturally falls out of the `as_components()` model, and with a bit of care should be easily attainable in all SDKs.  

- [ ] https://github.com/rerun-io/rerun/issues/3103
- [ ] https://github.com/rerun-io/rerun/issues/3159
- [ ] Extending builtin archetypes & components:
    - [x] Rust
    - [ ] C++
    - [ ] Python
- [ ] `custom_data` example
    - [x] Rust
    - [ ] C++
    - [ ] Python

nikolausWest, 2023-10-26:
Reminder to also update https://www.rerun.io/docs/howto/extend/custom-data when the C++ example for `custom_data` is done

teh-cmc, 2023-11-28:
Done! :partying_face:

closed 2023-11-28 by teh-cmc