rerun-io/rerun #3266: Investigate why `cargo check` is doing double-work

emilk, 2023-09-11:


Wumpf, 2023-09-22:
I isolated one pattern with deterministic & unnecessary rebuilds
* do a change
* `cargo rerun`
* `just py-build`
* `cargo rerun` <- will build winit again

Looking at it with `CARGO_LOG=cargo::core::compiler::fingerprint=trace` I found 
```
[2023-09-21T11:25:03Z INFO  cargo::core::compiler::fingerprint] dependency on `winit` is newer than we are 1695295464.541226352s > 1695295444.936536278s "/Users/andreas/.cargo/git/checkouts/egui-5e4507fa4153be06/d949eaf/crates/egui-winit"
[2023-09-21T11:25:03Z INFO  cargo::core::compiler::fingerprint] dependency on `winit` is newer than we are 1695295464.541226352s > 1695295445.096291254s "/Users/andreas/.cargo/git/checkouts/egui-5e4507fa4153be06/d949eaf/crates/eframe"
```

which seems to imply an issue with our current egui patches

emilk, 2023-09-26:
We have a lot of `build.rs` files following this pattern:

``` rs
fn main() {
    re_build_tools::rebuild_if_crate_changed("re_data_source");
    re_build_tools::export_env_vars();
}
```

we do this in order to get correct and up-to-date build info.

I wonder if we really need this though. Sure we want good build info for build from CI, but who really cares about the commit hash when you are building locally?

I added it originally for when working on the web-viewer, since it wasn't always obvious if the web-viewer that e.g. a notebook was serving was the latest built one (either due to bugs in the `build.rs` script that should trigger a rebuild of the web-viewer, or because of browser-side caching). In any case: this could be solves somehow else, e.g. with something opt-in.

emilk, 2023-09-26:
Running `cargo codegen` over and over will rebuild every times. That can't be right.

teh-cmc, 2023-09-26:
> Running `cargo codegen` over and over will rebuild `re_types_builder` every time. That can't be right.

It's the same underlying problem as the roundtrip test suite execution time regression:
- https://github.com/rerun-io/rerun/pull/3474

emilk, 2023-09-26:
This issue was opened before the PR that caused the bug fixed by #3474. There are more improvements to be had.

emilk, 2023-10-09:
Right now I'm leaning towards something like this:

* opt-in to most of our `build.rs`-shenanigans using an environment variable
* when unset, we don't show things like commit hash in the viewer (since it is unreliable without the environment variable)

This means you'd most often need to run `cargo codegen` manually, unless you turn on that environment variable.

emilk, 2023-10-12:
The problem that Andreas reported still remain on `main`:

```
> cargo rerun --version
…
> just py-build
…
> cargo rerun --version
   Compiling objc-sys v0.2.0-beta.2
   Compiling objc2-encode v2.0.0-pre.2
   Compiling objc2 v0.3.0-beta.3.patch-leaks.3
   Compiling winit v0.28.7
   Compiling accesskit_macos v0.7.1
   Compiling accesskit_winit v0.14.1
   Compiling egui-wgpu v0.23.0
   Compiling egui-winit v0.23.0
   Compiling eframe v0.23.0
   Compiling re_ui v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_ui)
   Compiling re_viewer_context v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_viewer_context)
   Compiling re_data_ui v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_data_ui)
   Compiling re_space_view v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_space_view)
   Compiling re_space_view_text_document v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_space_view_text_document)
   Compiling re_space_view_time_series v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_space_view_time_series)
   Compiling re_space_view_bar_chart v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_space_view_bar_chart)
   Compiling re_space_view_spatial v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_space_view_spatial)
   Compiling re_time_panel v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_time_panel)
   Compiling re_space_view_text_log v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_space_view_text_log)
   Compiling re_space_view_tensor v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_space_view_tensor)
   Compiling re_viewport v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_viewport)
   Compiling re_viewer v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/re_viewer)
   Compiling rerun v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/rerun)
   Compiling rerun-cli v0.10.0-alpha.6+dev (/Users/emilk/code/rerun/rerun/crates/rerun-cli)
    Finished dev [optimized + debuginfo] target(s) in 13.23s
     Running `target/debug/rerun --version`
```

This makes me wonder if the feature flags are changing somehow. `eframe` has some feature flags in `Cargo.toml` and some other in `crates/re_viewer/Cargo.toml` - I wonder what happens if I make them the same…

emilk, 2023-10-12:
`objc-sys` is the first crate being rebuilt. Maybe something is wrong in https://github.com/madsmtm/objc2/blob/master/crates/objc-sys/build.rs ? Or different feature flags are enabled on `objc-sys`?

PR #5197 (2024-02-15): Bump maturin to 1.14.0
### What

This should fix the recompilation issue related to the `MACOSX_DEPLOYMENT_TARGET` env var.

* Fixes #3266

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://app.rerun.io/pr/5197/index.html)
  * Using examples from latest `main` build: [app. [... cut]
commit 24dddad0 (2024-02-15): Bump maturin to 1.14.0 (#5197)

### What

This should fix the recompilation issue related to the
`MACOSX_DEPLOYMENT_TARGET` env var.

* Fixes #3266

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/5197/index.html)
* Using examples from latest `main` build:
 [... cut]
closed 2024-02-15 by emilk