rerun-io/rerun #3269: DSL for Analytics

jprochazk, 2023-09-11:
We need a file which lists all the events and their properties. The file needs to be readable by non-Rust users.

I think a DSL could work well here. We would use it to generate types and constructors/builders for the various events that we send, which would prevent the code from ever being out-of-sync with the file.

Here's an example of what it could look like:
```rust
/// Determines how the viewer was spawned, and in which environment it is running.
enum AppEnv {
  /// Running natively, spawned using the `rerun` CLI.
  RerunCli,
  
  /// Running natively, spawned in the Python SDK.
  PythonSdk,
  
  /// Running natively, spawned in the Rust SDK.
  RustSdk,
  
  /// Running on the web.
  WebViewer,
}

/// Determines how data is getting to the viewer.
enum DataSource {
  /// Reading data from a TCP socket.
  TcpServer,
  
  /// Reading data from a local file.
  File,
  
  /// Reading data from a remote file over HTTP.
  Http,
  
  /// Reading data logged live from one of the SDKs.
  Sdk,
  
  /// Reading data from a WebSocket connection.
  WsClient,
}

define_analytics! {
  /// Global properties sent with every event.
  global {
    /// Environment in which the viewer is running
    app_env: AppEnv,

    /// ID of the current session, changes each time
    /// the viewer is opened
    session_id: String,
  }

  /// Sent when a user opens a recording in the viewer.
  event OpenRecording {
    /// Unique ID of the recording
    recording_id: String,

    /// How the data is being sent to the viewer
    data_source: DataSource,

    /// Where the data is coming from
    store_source: String,

    /// Version of Python the SDK is running on
    python_version: String,
  };
}
```

This would generate an `Analytics` struct which would accept the global properties in the constructor,
and have methods for creating each defined `event`. The events would use the builder pattern, so that we can have optional properties, and the event itself can be built up as needed in the code:

```rust
analytics
  .open_recording(recording_id, data_source, store_source)
  // here we have a mutually exclusive `rust_version + llvm_version` or `python_version`
  // appending one property removes the others, and the event may not be finished without either
  .python_version(python_version)
  .send();
```

This would replace our current manually-defined events on the also manually-defined `Analytics` struct.

emilk, 2023-09-11:
Let's keep it simple:

```rs
trait AnalyticsEvent {
    NAME: &'static str,

    fn into_event(self) -> Event
}

// All the fields map 1-to-1 to the final fields

pub struct CrashPanicEvent {
    /// …
    pub callstack: String,

    /// …
    pub message: String,

    /// …
    pub file_line: String,
}

…
```

jprochazk, 2024-02-22:
Some notes:

We have a lot of "global" properties that are appended to every event. Currently those are done ad-hoc via `register` calls on `ViewerAnalytics`, placing the properties in an untyped `HashMap<String, Property>`.
We'll move these into a `Context` struct, which will live in `ViewerAnalytics`.

Every property that is _not_ registered globally will be part of a specific event struct. We don't have many of these.

PR #5267 (2024-02-27): Strongly-typed analytics
### What

- Closes #3269 
- Closes #3271 

We want analytics to be easier to grok at a glance, both for our own and our users' sake. Events and their properties should be defined in plain structs with as little other boilerplate around them as possible. Anyone who's ever read a statically-typed language with a `struct`-like concept should be able to understand them.

This will also make it easier for us to track our own analytics events. What kinds are there, and what properties do they have, all defined in one place. The only exception is the events we send on our web [... cut]
commit 1ff167c4 (2024-02-27): Strongly-typed analytics (#5267)

### What

- Closes #3269 
- Closes #3271 

We want analytics to be easier to grok at a glance, both for our own and
our users' sake. Events and their properties should be defined in plain
structs with as little other boilerplate around them as possible. Anyone
who's ever read a statically-typed language with a `struct`-like concept
should be able to understand them.

This will also make it easier for us to track our own analytics events.
What kinds are there, and what properties do they have, all defined in
one place. The only exception is the events we send o [... cut]
closed 2024-02-27 by jprochazk