rerun-io/rerun #3270: new api crashes when archetype num instances is smaller than any of the component arrays

Wumpf, 2023-09-11:
Simple crash repro from python
`rr2.log("crash", rr2.Points2D([0, 0], instance_keys=[0, 1]))`

teh-cmc, 2023-09-11:
```
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: WrongNumberOfInstances { entity_path: "crash", component: "rerun.instance_key", expected_num_instances: 1, num_instances: 2 }', /home/cmc/dev/rerun-io/rerun2/crates/re_log_types/src/data_row.rs:315:84
stack backtrace:
   0: rust_begin_unwind
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/std/src/panicking.rs:593:5
   1: core::panicking::panic_fmt
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/core/src/panicking.rs:67:14
   2: core::result::unwrap_failed
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/core/src/result.rs:1651:5
   3: core::result::Result<T,E>::unwrap
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/core/src/result.rs:1076:23
   4: re_log_types::data_row::DataRow::from_cells
             at /home/cmc/dev/rerun-io/rerun2/crates/re_log_types/src/data_row.rs:315:9
   5: rerun_bindings::arrow::build_data_row_from_components
             at /home/cmc/dev/rerun-io/rerun2/rerun_py/src/arrow.rs:102:15
   6: rerun_bindings::python_bridge::log_arrow_msg
             at /home/cmc/dev/rerun-io/rerun2/rerun_py/src/python_bridge.rs:1219:15
   7: rerun_bindings::python_bridge::_::__pyfunction_log_arrow_msg
             at /home/cmc/dev/rerun-io/rerun2/rerun_py/src/python_bridge.rs:1197:1
   8: pyo3::impl_::trampoline::cfunction_with_keywords::{{closure}}
             at /home/cmc/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pyo3-0.19.0/src/impl_/trampoline.rs:41:35
   9: pyo3::impl_::trampoline::trampoline_inner::{{closure}}
             at /home/cmc/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pyo3-0.19.0/src/impl_/trampoline.rs:181:54
  10: std::panicking::try::do_call
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/std/src/panicking.rs:500:40
  11: std::panicking::try
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/std/src/panicking.rs:464:19
  12: std::panic::catch_unwind
             at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/std/src/panic.rs:142:14
  13: pyo3::impl_::trampoline::trampoline_inner
             at /home/cmc/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pyo3-0.19.0/src/impl_/trampoline.rs:181:9
  14: pyo3::impl_::trampoline::cfunction_with_keywords
             at /home/cmc/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pyo3-0.19.0/src/impl_/trampoline.rs:41:13
  15: rerun_bindings::python_bridge::_::<impl rerun_bindings::python_bridge::log_arrow_msg::MakeDef>::DEF::trampoline
             at /home/cmc/dev/rerun-io/rerun2/rerun_py/src/python_bridge.rs:1197:1
  16: <unknown>
  17: _PyObject_MakeTpCall
  18: _PyEval_EvalFrameDefault
  19: _PyEval_EvalCodeWithName
  20: _PyFunction_Vectorcall
  21: _PyEval_EvalFrameDefault
  22: _PyEval_EvalCodeWithName
  23: PyEval_EvalCodeEx
  24: PyEval_EvalCode
  25: <unknown>
  26: <unknown>
  27: <unknown>
  28: PyRun_SimpleFileExFlags
  29: Py_RunMain
  30: Py_BytesMain
  31: <unknown>
  32: __libc_start_main
  33: _start
```

emilk, 2023-09-22:
It turns out we shouldn't put `unwrap`:s in our code

commit 3e222ac7 (2023-09-22): Add `_or_panic` suffix to functions that panics on error

…and stop using them in a bunch of places.

* Closes #3270

We cannot do `.unwrap()` in functions with innocuous names.
We will get crashes.
Use `Result`.
PR #3406 (2023-09-23): Fix crash on mismatch component lengths
* Closes #3270

Adds `_or_panic` suffix to functions that panics on error and stops using them in a bunch of places.

We cannot do `.unwrap()` in functions with innocuous names. We will get crashes. Use `Result`.

Next step is to move away from these horrible `_or_panic` functions, because we don't want the process to die.

If you add a `.unwrap()` explain _why it can't happen_.
Any naked (unexplained) `.unwrap()` in our code is an implicit TODO that should be fixed.
Ideally we should enable `clippy::unwrap_used` on our code-base.

### Checklist
* [x] I  [... cut]
commit 2e9df4ca (2023-09-23): Fix crash on mismatch component lengths (#3406)

* Closes #3270

Adds `_or_panic` suffix to functions that panics on error and stops
using them in a bunch of places.

We cannot do `.unwrap()` in functions with innocuous names. We will get
crashes. Use `Result`.

Next step is to move away from these horrible `_or_panic` functions,
because we don't want the process to die.

If you add a `.unwrap()` explain _why it can't happen_.
Any naked (unexplained) `.unwrap()` in our code is an implicit TODO that
should be fixed.
Ideally we should enable `clippy::unwrap_used` on our code-base.

### Checklist [... cut]
closed 2023-09-23 by emilk