rerun-io/rerun #3273: Update general documentation for new api

nikolausWest, 2023-09-11:
Update all our general (non-sdk specific) documentation pages to use the new api:

### Getting Started
- [ ] Python Quick Start
- [ ] Rust Quick Start
- [ ] Logging Data in Python
- [ ] Logging Data in Rust

### How-To Guides
- [ ] Using Rerun in Notebooks
- [ ] using Rerun with ROS 2

### Concepts
- [ ] Entities and Components
- [ ] The Entity Path Hierarchy
- [ ] Spaces and Transforms
- [ ] Events and Timelines
- [ ] Annotation Context

nikolausWest, 2023-09-25:
Running notes that we can fill in while porting all examples etc: https://www.notion.so/rerunio/0-9-migration-notes-2b3debe9947f47a38428267b2da4cc1b?pvs=4

We'll then use those notes to create the migration guide

emilk, 2023-09-25:
Blocked by
* https://github.com/rerun-io/rerun/issues/3236

Wumpf, 2023-10-04:
there's a bunch of follow-up comments on the concepts

closed 2023-10-05 by emilk