rerun-io/rerun #3279: provide `empty()` method on Rust archetype implementations to log empty

Wumpf, 2023-09-11:
It's cumbersome right now to log an empty set of things:
```rust
rec.log("rects_test/rects", &Boxes2D::new([HalfExtents2D; 0]))?;
```

this should be instead
```rust
rec.log("rects_test/rects", &Boxes2D::empty())?;
```

nikolausWest, 2023-09-11:
C++ and Python too?

teh-cmc, 2023-09-12:
@jleibs had an intesresting question:
> Is this different from Clear?

The answer is yes for three reasons:
1. `clear()` has optional recursiveness
2. `clear()` has the out-of-order / pendingness thing (which I'm not sure I understand what that's expected to do anymore)
3. and finally `clear()` doesnt stop at a specific archetype: if the entity has 10 archetypes logged, it clears all of them

Wumpf, 2023-09-12:
In C++ and Python we have harder to avoid default construction which I think is more suited there for creating empty archetypes 🤔

emilk, 2023-09-19:
We should make `empty()` set all the components to empty lists so that they overwrite everything. We should probably make the default constructor do the same.

emilk, 2023-09-20:
* Closing in favor of https://github.com/rerun-io/rerun/issues/3381

closed 2023-09-20 by emilk