rerun-io/rerun #3283: Can't `log_segmentation_image` if 1px wide

pd-nisse, 2023-09-12:
**Describe the bug**
I have a camera sensor that is just 1 pixel wide. When I'm trying to log the segmentation image of shape `(4100, 1, 1)` , I receive error `TypeError: Expected single channel image, got array of shape (4100, 1, 1). Can't interpret as segmentation image.`

**To Reproduce**
Steps to reproduce the behavior:
`rr.log_segmentation_image(entity_path="image", image=np.ones(shape=(4100, 1, 1).astype(np.uint8))`

**Expected behavior**
It just logs the segmentation mask.

**Screenshots**
I know it's thin, but it seems like a legitimate use case to have 1 pixel wide / high images.
[image]

**Backtrace**
```
    rr.log_segmentation_image(entity_path=f"{image_ref}/{annotation_identifier.name}", image=class_ids)
  File "/Users/nisseknudsen/.pyenv/versions/paralleldomain_internal/lib/python3.10/site-packages/rerun_sdk/rerun/log/log_decorator.py", line 48, in wrapper
    return func(*args, **kwargs)
  File "/Users/nisseknudsen/.pyenv/versions/paralleldomain_internal/lib/python3.10/site-packages/rerun_sdk/rerun/log/image.py", line 257, in log_segmentation_image
    _send_warning(
  File "/Users/nisseknudsen/.pyenv/versions/paralleldomain_internal/lib/python3.10/site-packages/rerun_sdk/rerun/log/error_utils.py", line 35, in _send_warning
    raise TypeError(message)
TypeError: Expected single channel image, got array of shape (4100, 1, 1). Can't interpret as segmentation image.
```

**Desktop (please complete the following information):**
 - OS: macOS Ventura 13.5.2

**Rerun version**
`0.8.2`

**Additional context**
It seems in `rerun/log/image.py` , every dimension that is `1` is being discarded as empty.
[image]

pd-nisse, 2023-09-12:
Potentially related:
When I log 2pixels wide image, `rerun` does not thrown an exception, although in-app I do see "Condition Failed" error in several Tensor fields. 
[image]

emilk, 2023-09-12:
This is mostly fixed on `main`, except that `rr2.log("narrow", rr2.SegmentationImage(np.ones(shape=(10, 1, 1)).astype(np.uint8)))` is misinterpreted as a bar-chart 🤦  This will be fixed once we migrate to using indicator components for everything though.

emilk, 2023-09-24:
This is likely fixed on `main` now, but let's check before we close it

jleibs, 2023-09-25:
The direct logging/viewing restriction has been resolved (i.e. you can now use a 200x1 image), but the 4000x2 issue remains. Closing in favor of: https://github.com/rerun-io/rerun/issues/3430 remains.

closed 2023-09-25 by jleibs