rerun-io/rerun #3297: TranslationAndMat3x3 defaults to zero matrix instead of identity

roym899, 2023-09-12:
**To Reproduce**
Compare
```
import numpy as np
import rerun as rr
import rerun.experimental as rr2
from rerun.experimental import dt as rrd

rr.init("depth_camera", spawn=True)
rr2.log(
    "world/rgbd_camera",
    rr2.Transform3D(rrd.TranslationAndMat3x3(matrix=np.eye(3))),
)
rr.log_pinhole("world/rgbd_camera", height=480, width=640, focal_length_px=500.0)
```
with and without the matrix argument.

**Expected behavior**
Default should be the identity.

**Screenshots**
With argument:
[image]
Without argument:
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
```
rerun_py 0.9.0-alpha.4 [rustc 1.72.0 (5680fa18f 2023-08-23), LLVM 16.0.5] x86_64-unknown-linux-gnu 7db7b54, built 2023-09-12T08:50:21Z
```

emilk, 2023-09-12:
Good catch!

teh-cmc, 2023-09-13:
The matrix is optional and should serialize as `null` when left unspecified... so the issue is probably query-side and affect all SDKs?

emilk, 2023-09-13:
Good point, and the cause of the problem is likely then the auto-derive of `Default` for `Mat3x3`. Sounds like a very easy fix then, but we should also check if we do the same bad auto-derive for other types (e.g. quaternions)

closed 2023-09-13 by emilk