rerun-io/rerun #3298: Logging boolean tensors not supported

roym899, 2023-09-12:
**Describe the annoyance**
It'd be nice if boolean tensors could be logged (and show up as black and white masks in Rerun).

**To Reproduce**
```
import numpy as np
import rerun as rr
import rerun.experimental as rr2

rr.init("mask", spawn=True)
mask = np.random.rand(100, 100) > 0.5
rr2.log("mask", rr2.Tensor(mask))
```
gives
```
AssertionError: Failed to find <class 'numpy.bool_'> in f{<class 'numpy.uint8'>: 'U8', <class 'numpy.uint16'>: 'U16', <class 'numpy.uint32'>: 'U32', <class 'numpy.uint64'>: 'U64', <class 'numpy.int8'>: 'I8', <class 'numpy.int16'>: 'I16', <class 'numpy.int32'>: 'I32', <class 'numpy.int64'>: 'I64', <class 'numpy.float16'>: 'F16', <class 'numpy.float32'>: 'F32', <class 'numpy.float64'>: 'F64'}
```

**Expected behavior**
Show a black and white mask in the Rerun viewer.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
```
rerun_py 0.9.0-alpha.4 [rustc 1.72.0 (5680fa18f 2023-08-23), LLVM 16.0.5] x86_64-unknown-linux-gnu 7db7b54, built 2023-09-12T08:50:21Z
```

emilk, 2023-09-12:
Good point - this should be pretty easy to implement _if_ we start with the naive one-byte-per-bool implementation. A proper one-bit-per-bool implementation shouldn't be _that_ much harder though. I think arrow2 has a buffer for that.

roym899, 2024-04-09:
This also applies to SegmentationImages. Right now an extra `.astype(np.uint8)` is needed.
```python
binary_segmentation_mask = binary_segmentation_probabilities > 0.5
rr.log("video/mask", rr.SegmentationImage(binary_segmentation_mask.astype(np.uint8)))
```