rerun-io/rerun #3330: [C++] remove `with_singular` and consider variadic `with_plural` on archetypes

Wumpf, 2023-09-15:
Singular is inconsistent with how we roll on other platforms.
Passing several custom arguments could be done better with a variadic template, so we go from
```
.with_labels({"red", "green", "blue"})
```
to 
```
.with_labels("red", "green", "blue")
```

should still support passing in vectors though! This part is related to #3050

emilk, 2023-10-09:
I think it is important that we still support passing in a `std::vector` of labels (or ideally any container compatible with `std::begin/std::end`), and I'm not sure variadic templates is compatible with this.

I don't think it is inherently bad to distinguish between singular and plural. `Points3D(…).with_color(0xff00ff)` reads pretty good to me in both C++ and Rust.

Wumpf, 2023-10-12:
Settled by #3791: ComponentBatches can be created from single and multiple components and the way this is designed it now doesn't really make sense to have singualr versions

closed 2023-10-12 by Wumpf