rerun-io/rerun #3341: Decide if archetypes logging APIs should log non-provided components as empty-lists

jleibs, 2023-09-15:
On a few occasions we have proposed that logging via the new archetype APIs should always log every component, implicitly clearing any field which is not provided.

The main benefit of this is two-fold:
1. It avoids needing to explain join-semantics to users that are only using the Archetype APIs. By default, entities will always show up "as logged" and never take on values from prior log calls.
1. It defaults to a behavior which is much more performant since the latest-at temporal search and the join implementation are both more costly than when all values of an archetype are found in the same row.

However, this would be a departure from the historical behavior, so for backwards compatibility, so far we have avoided making this change.

emilk, 2023-09-20:
There are also benefits from a user standpoint:

* `log(ent_path, Points3d::empty())` will clear the given entity, which seems logical
* `log(ent_path, Points3d::new(poisitoons))` will ignore earlier colors, which seems logical

emilk, 2023-09-20:
We've all converged on the following API:

``` rs
// We do NOT implement `Default` for archetypes.
// users need to explcitly use `clear` or `update`
struct Point3d {
   positions: Optional<Vec<Vec3>>,
   color: Optional<Vec<Color>>,
   radii: Optional<Vec<f32>>,
}

impl Points3d {
    /// Clear all components of this archetype to the empty list.
    fn clear() -> Self {
        Self {
            positions: Some(vec![]),
            color: Some(vec![]),
            radii: Some(vec![]),
        }
    }
    
    /// Create a new `Point3d` point cloud with some positions,
    /// and with emtpy lists for all other archetypes.
    fn new(positions: Vec<Vec3>) -> Self {
        Self {
            positions: Some(positions),
            ..Self::clear()
        }
    }
    
    /// An empty `Point3d` point cloud, with `None` for each component.
    fn update() -> Self {
        Self {
            positions: None,
            color: None,
            radii: None,
        }
    }
}

fn main() {
    // Log a point cloud:
    log("points", Point3d::new(…).with_colors(…));
    
    log("points", Clear{}); // will clear all components of all archetypes
    log("points", Point3d::clear()); // will clear this archetype of the entity
    log("points", Point3d::new(positions).with_colors(…)); // overwrites everything
    log("points", Point3d::clear().with_colors(…)); // overwrites everything
    log("points", Point3d::update().with_colors(…)); // overwrites only colors
}
```

Python:
```
log("points", Clear()); // will clear all components of all archetypes
log("points", Point3d.clear()); // will clear this archetype of the entity
log("points", Point3d(positions, colors=[…], radii=None); // overwrites everything, even radii
log("points", Point3d.update(colors=[…], radii=None)); // overwrites only colors and radii
```

### Necessary changes
* All components are `Options`, even the required ones
* When querying, an empty lists should be treated the same as a missing components
* In the UI we should probably hide empty components in the streams view

### Component lists implications
A component list (e.g. `radii`) can have one of the four special values:

* null/missing
* 0 length ("empty")
* 1 length ("splat")
* 1 < N length

* `null` components aren't logged
* empty components are logged
* when querying the store, missing components are treated the same as empty components

emilk, 2023-09-20:
* Closing in favor of https://github.com/rerun-io/rerun/issues/3381