rerun-io/rerun #3342: When no heuristics match, still pick a fallback view if possible

jleibs, 2023-09-15:
As of https://github.com/rerun-io/rerun/pull/3323 our space view part systems now use indicator components to filter the heuristics. This means if we *just* log via the new `as_components` APIs, we don't get any views. Patching in indicators is still a bit painful, so it would be nice if we had sane fallback behavior.

emilk, 2023-09-25:
Let's do it for 0.9 if it is easy

Wumpf, 2023-09-27:
The basics of implementing this are easy as expected. Instead of discarding entities based on heuristics from the `EntitiesPerSystemPerClass` list immediately we can do something like this:

```
pub enum ViewPartHeuristicFilterResult {
    /// The entity path has all required components for the part system and passes the heuristic filter.
    ///
    /// This means that the PartSystem will with very high likelihood display something meaningful for this entity.
    Passes,

    /// The entity path has all *required* components for the part system, but fails the heuristic filter.
    ///
    /// This typically means that it's either lacking an indicator component or
    /// has some property that make not possible to be displayed by the given part system.
    Fails,
}

pub type EntitiesPerSystem =
    IntMap<ViewSystemName, IntMap<EntityPath, ViewPartHeuristicFilterResult>>;

pub type EntitiesPerSystemPerClass = IntMap<SpaceViewClassName, EntitiesPerSystem>;
```

We then can use the heuristic result to spawn spaceviews where the heuristic passes or just the first one where it doesn't. (Would be nice to have a better heuristic than "the first")
For this we'd likely forward this new piece of information to `SpaceViewClass::auto_spawn_heuristic`

The tricky part that requires a bit deeper changes is how to add entities to an _existing_ space view: The problem is that we do this in a localized manner right now. Each Space View looks at the list of entities its systems would be interested in and then decides whether to add it. Previously, this list was already filtered by the heuristic. With the above altered datastructure however we need to make a choice depending on _whether another Space View_ was better suited for an entity that *didn't* pass those heuristics. This means that the decision which incoming entities to add to which existing Space View becomes a central one and not one we can solve by looking at one (existing) Space View at a time.

➡️ In the near future the heuristic pass needs pass need to be structured like this (not too unsimilar to now):
* compute `EntitiesPerSystemPerClass` with heuristic information (note that this is affected by the timeline position and incoming data)
* if enabled, add new space views, using an enhanced `SpaceViewClass::auto_spawn_heuristic`. TODO: Ideally we rewrite this step to not yet populate entities so this can be done in the next step?
* for all entities that aren't in any space view yet at all determine where they should go among the lists of Space Views with the auto add entities flag enabled

The change for the last step is a bit too much and too dangerous for 0.9. Delayed to 0.10

emilk, 2023-10-09:
Maybe related:
* https://github.com/rerun-io/rerun/issues/3709

PR #4874 (2024-01-28): Improved automatic view creation heuristic, major speedup for scenes with many entities
### What

What space views get spawned is no longer determined by a centralized heuristic, but instead by one distributed to each space view class. Each of those heuristics can make much better use of existing information provided by store subscribers.

* Fixes most of #4388
   *  a few things on the tracking issue still left to investigate
* Fixes #3342 
   * more "obsoleted" arguably: We originally had a hack that 1d tensor logging would display a bar chart. That's not actually desirable. There is still t [... cut]
commit 15cd60d8 (2024-01-28): Improved automatic view creation heuristic, major speedup for scenes with many entities (#4874)

### What

What space views get spawned is no longer determined by a centralized
heuristic, but instead by one distributed to each space view class. Each
of those heuristics can make much better use of existing information
provided by store subscribers.

* Fixes most of #4388
   *  a few things on the tracking issue still left to investigate
* Fixes #3342 
* more "obsoleted" arguably: We originally had a hack that 1d tensor
logging would display a bar chart. That's not actually desirable. There
is s [... cut]
closed 2024-01-28 by Wumpf