rerun-io/rerun #3384: Use `enum` in codegen to specify constant-constants

jleibs, 2023-09-20:
Our `enum`s currently get converted into classes that implement arrow unions, which I don't believe is correct. The more common pattern of using enums in fbs files, for example:

```
enum ViewDir: byte {
    Up = 1,
    Down = 2,
    Right = 3,
    Left = 4,
    Forward = 5,
    Back = 6,
}
```

Should map to an appropriate enum of constants in each language.

Wumpf, 2023-09-20:
would also be useful for log level strings!

abey79, 2023-12-18:
...and for blueprint container kind.

emilk, 2024-01-11:
There are two ways we can encode the enums: using names ("Up", "Down", …), or using integers (1, 2, …).

I think using the names is superior, as it means the arrow data is self-descriptive. However, it means renaming an enum variant is a breaking change (just like for the name of a struct field).

I believe the best way to encode these in Arrow is by using a Sparse Union with the NULL type for each field. This means only 8 bits should be required for each value.

closed 2024-02-29 by emilk