rerun-io/rerun #3399: Hovering the barchart legend causes the chart to disappear

jleibs, 2023-09-21:
**Describe the bug**
When moving the mouse over the legend, the chart disappears.

**To Reproduce**
Go to: https://demo.rerun.io/version/0.8.2/examples/plots/
Move mouse over the legend that dsays "bar chart"

**Expected behavior**
The chart shouldn't disappear. Maybe it should even become more visible rather than less visible.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.10
 
**Rerun version**
0.8.2 and main.

emilk, 2023-09-25:
Probably something to fix in `egui_plot`

emilk, 2023-09-25:
Will be fixed when we update to egui with:
* https://github.com/emilk/egui/pull/3387

closed 2023-09-25 by emilk