rerun-io/rerun #3402: Force users to spell out the components in archetype constructors

emilk, 2023-09-22:
Currently this is valid code:

```py
rr.log("base", rr.Arrows3D([0, 1, 0]))
```

As a user I would find this profoundly confusing. What are these three numbers - are they the base of the arrow, or its tip? (answer: neither!).

This would be much better if the user was forced to write out `rr.log("base", rr.Arrows3D(vector=[0, 1, 0]))`.
(In this case, I would also vote for making `origin` mandatory).

Furthermore, I think in all our examples we should include the origin for our arrows.

The same would go for Rust and C++ - `Arrows3D::from_vector(…).with_origin(…)`.

nikolausWest, 2023-09-22:
I think forcing named arguments here is fine but I don't think we should make origin mandatory.

teh-cmc, 2023-09-22:
Agreed with @nikolausWest.

Making `origin` mandatory would also introduce the concept of archetypes with multiple required components which we're really not ready to deal with for 0.9.

PR #3411 (2023-09-24): Enforce named constructor (argument) for Arrows3D
### What

* Fixes #3402
-> as discussed, Arrows3D is the only instance where we want to change this right now. Note that we already deal with Boxes2D/Boxes3D in a similar fashion

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3411) (if applicable)

- [PR Build Su [... cut]
commit 8909fcd6 (2023-09-24): Enforce named constructor (argument) for Arrows3D (#3411)

### What

* Fixes #3402
-> as discussed, Arrows3D is the only instance where we want to change
this right now. Note that we already deal with Boxes2D/Boxes3D in a
similar fashion

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3411) (if
applicable)

- [PR [... cut]
closed 2023-09-24 by jleibs