rerun-io/rerun #3418: `rr2.Clear` too difficult to understand

roym899, 2023-09-23:
**Describe the annoyance**
Maybe this is just not fully done (I saw there are some TODOs in the docstring still, however, I thought to still bring it up), but for `rr2.Clear(settings)`, I only see `ClearSettingsArrayLike` as the type hint for the settings argument, however, the code example just passes True / False (not documented sufficiently right now). `ClearSettingsArrayLike` is quite uninformative.

If it was defined as `rr2.Clear(recursive: bool)` it would be much clearer how to use it, just from the function signature.

Also, right now it does not work without arguments. `rr2.Clear(None)` behavior is not documented. Could it be `rr2.Clear(recursive: bool=False)`?

teh-cmc, 2023-09-23:
The python API is actually missing some extensions for some reason (you can see them in action in the Rust or C++ examples).

Once those extensions are implemented, this would become:
```python
rr2.log("my_entity", rr2.Clear.flat())
rr2.log("my_entity", rr2.Clear.recursive())
```

Is that better?

roym899, 2023-09-23:
Any reason why this pattern over `rr2.Clear(recursive: bool=False)`? 

I wouldn't be very opposed to it, but would prefer the latter for the reasons below.

1. It feels a little inconsistent with the other archetypes. Maybe I'm wrong here, but would this be the only archetype that uses factory methods? (feels like that could also make sense for some other archetypes, such as Transform3D)
2. When `flat()` and `recursive()` are available, would there still be the option to create it as `rr2.Clear(...)` or would that somehow be disabled / give an error? 
3. Without knowing of the existence of factory methods, I'd probably default to using the default way of creating the class in Python, i.e., I'd try to use `rr2.Clear(...)` and see what arguments it takes, instead of checking whether there are factory methods available to use

teh-cmc, 2023-09-25:
Thanks for the detailed write up. `rr2.Clear(recursive: bool=False)` should be doable, let's see what we can do..!

closed 2023-09-25 by emilk