rerun-io/rerun #3420: World up-axis not respected under various conditions

roym899, 2023-09-24:
**Describe the bug**
It seems like after setting the up-axis using view coordinates the Rerun viewer has to be restarted for this to actually have an effect.

This seems to be the case when:
- rotating the camera around its principal axes (i.e., middle-mouse rotate)
- switching between recordings with different up-axis
- logging again, but with different up-axis

**To Reproduce**
```python
import numpy as np
import rerun as rr
import rerun.experimental as rr2

rr.init("Up-axis Test", spawn=True)
rr2.log("world", rr2.ViewCoordinates.RUB, timeless=True)
rr2.log(
    "world/points",
    rr2.Points3D(np.random.rand(100, 3))
)
```
Run this twice and replace RUB with RDF. The second recording will say up is -Y, but the view shows +Y as the up-axis still.

**Expected behavior**
3D view should always respect the set up-axis.

After camera rotation it's a bit less obvious what should happen. But still rotating around the up-axis seems better then whatever it rotates around right now (?). Generally, the rotation when no up-axis has been set feels unintuitive, but that's a separate issue.

**Desktop (please complete the following information):**
 - OS Ubuntu 20.04
 
**Rerun version**
```
rerun_py 0.9.0-alpha.4 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu 2e9df4c, built 2023-09-23T09:59:35Z
```

roym899, 2023-09-24:
I think once I figured out that it's fixed by double clicking it's less of an issue.

emilk, 2023-10-09:
Very related:
* https://github.com/rerun-io/rerun/issues/3539

commit 1e5f9332 (2024-02-15): Improve the orbit eye to always maintain an up-axis (#5193)

* Continuing the work in https://github.com/rerun-io/rerun/pull/3817

### What
- The orbit eye now always maintains an up axis internally (+Z by
default, if there is no `ViewCoordinates`).
- When a user rolls the camera (middle mouse drag), we now change the
internal up-axis, which makes rolling then orbiting a much better
experience as you don't end up in a weird state with no up axis.
- The orbit eye center visualization now indicates the current up-axis

https://github.com/rerun-io/rerun/assets/1148717/daeb1e57-8fe3-4ac9-84fc-c673 [... cut]
closed 2024-02-15 by emilk