rerun-io/rerun #3421: No way to adjust Recordings / Blueprint size in left-hand panel

roym899, 2023-09-24:
**Describe the annoyance**
There's no way to adjust how much space Recordings and Blueprint take on the left hand side. I think when having a lot of entities, it might often be desirable to have Blueprints take more space than it currently does. 

I think one would expect to have a resize option popping up when hovering over the top of Blueprint as in this video

https://github.com/rerun-io/rerun/assets/9785832/e404b60d-1619-4583-b713-f4a71be01f7a

```
rerun_py 0.9.0-alpha.4 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu 2e9df4c, built 2023-09-23T09:59:35Z
```

emilk, 2023-09-24:
The video isn't working for me

roym899, 2023-09-24:
Hm, I reencoded it with `yuv420p` pixel format. Not sure if that was the problem though.

emilk, 2023-09-25:
That works now - thanks!

emilk, 2023-09-25:
I think I need to add something like a `SplitPanel` to egui for this

closed 2024-08-14 by teh-cmc