rerun-io/rerun #3433: Archetype doc strings should explain parameters

nikolausWest, 2023-09-25:
Currently, the docstrings for archetypes in python don't include documentation of the parameters of the init methods. The types are not descriptive enough which leaves the API's very hard to understand.

For example, this is quite hard to use:
[image]

emilk, 2023-09-25:
Our python API docs generator should fail if we are missing docstrings for parameters

jleibs, 2023-09-25:
There's an open question as to whether these docs should go in the class docs or the __init__ function docs. Let's go with whatever gives the best IDE experience in vscode + pycharm.

Wumpf, 2023-09-25:
Related to #3423

PR #3480 (2023-09-27): Generate documented __init__ methods and better document custom __init__ methods
### What

* Fixes #3423
* Fixes #3433

---

* We now always emit `__init__` methods in python alongside with a doc string.
* Improved docstring for manual `__init__` methods
* fixed lots of type issues that bubbled up by doing the above

LSP is now very happy with me on pretty much all types:

manual `__init__`:
[image]

generated:
[image]

mouse hovering the type itself:
[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [ [... cut]
commit 6d1edffb (2023-09-27): Generate documented __init__ methods and better document custom __init__ methods (#3480)

### What

* Fixes #3423
* Fixes #3433

---

* We now always emit `__init__` methods in python alongside with a doc
string.
* Improved docstring for manual `__init__` methods
* fixed lots of type issues that bubbled up by doing the above

LSP is now very happy with me on pretty much all types:

manual `__init__`:

[image]

generated:

[image]

mouse hovering the type itself:

[image]

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.m [... cut]