rerun-io/rerun #3465: Make whether labels are shown a configurable property

jleibs, 2023-09-25:
We currently have hard-coded thresholds for whether to show labels.

Points and lines = 10
Boxes = 20

These values should just govern a threshold, but whether we show labels our not should be configurable via the UI.

Wumpf, 2024-07-03:
Related to
* https://github.com/rerun-io/rerun/issues/4266

Wumpf, 2024-07-23:
Adding a bit of context since this comes up more and more often now:
The limit is in place for usecases where labels should show only on hover, most typically this is an issue with large point clouds where it wouldn't make sense to show all labels at once.

I propose to introduce a `hide labels` boolean component that internally has a visualizer-dependent fallback that accounts for this very issue. This may not be the final solution we want to arrive here (e.g. independent archetype is still on the table https://github.com/rerun-io/rerun/issues/4266), but it would be fairly easy to implement and solves the problem at hand that various users run into!

jleibs, 2024-07-24:
This sounds great -- let's do it.

emilk, 2024-08-05:
Maybe something for @kpreid

PR #7266 (2024-08-28): Pass `ShowLabels` default through component fallback provider system.
### What

* Followup to #7249
* Fixes #3465

The default value for the `ShowLabels` component now passes through `ComponentFallbackProvider`, so the viewer UI will properly show the fallback value derived from the number of instances.

Further possible work in this area would be to define “number of instances” as something that can be generically fetched based on an archetype, rather than requiring each visualizer to communicate how it counts that.

### Checklist
* [X] I have read and agree to [Contributor Guide](https://git [... cut]
commit dc462a8f (2024-08-28): Pass `ShowLabels` default through component fallback provider system. (#7266)

### What

* Followup to #7249
* Fixes #3465

The default value for the `ShowLabels` component now passes through
`ComponentFallbackProvider`, so the viewer UI will properly show the
fallback value derived from the number of instances.

Further possible work in this area would be to define “number of
instances” as something that can be generically fetched based on an
archetype, rather than requiring each visualizer to communicate how it
counts that.

### Checklist
* [X] I have read and agree to [Contributor
Guide](ht [... cut]
commit cbfad294 (2024-08-28): Pass `ShowLabels` default through component fallback provider system. (#7266)

### What

* Followup to #7249
* Fixes #3465

The default value for the `ShowLabels` component now passes through
`ComponentFallbackProvider`, so the viewer UI will properly show the
fallback value derived from the number of instances.

Further possible work in this area would be to define “number of
instances” as something that can be generically fetched based on an
archetype, rather than requiring each visualizer to communicate how it
counts that.

### Checklist
* [X] I have read and agree to [Contributor
Guide](ht [... cut]
closed 2024-08-28 by jleibs