rerun-io/rerun #3473: `MeshProperties` required to create mesh

roym899, 2023-09-26:
**Describe the annoyance**
Logging meshes now requires to create a `MeshProperties` object to set the vertex indices. This requires an extra import, and I need to remember what part of the mesh belongs into mesh properties and what not.

Feels like all the other arguments that `Mesh3D` takes are also mesh properties, so it feels a bit arbitrary that the indices are wrapped with a separate properties object.

Any reason for this split?

nikolausWest, 2023-09-26:
Related: maybe we should have the rule of thumb that all objects that are expected arguments of an Archetype should go in the global namespace?

teh-cmc, 2023-09-26:
> Feels like all the other arguments that Mesh3D takes are also mesh properties

The `Mesh3D` archetype doesn't contain _mesh_ properties but rather _vertex_ properties. The distinction is important because there is no relationship between the number of vertices vs. the number of indices, which would violate our instance count invariants if they were both present together at the top-level (whether we want to relax this invariant in the future remains to be determined).
https://github.com/rerun-io/rerun/blob/main/crates/re_types/definitions/rerun/archetypes/mesh3d.fbs#L29

Wumpf, 2023-09-26:
But meanwhile we can crate an `__init__` method that takes `indices=` as a named argument. Seems like a good idea in any case, no?

roym899, 2023-09-26:
Ah, that makes sense @teh-cmc, still a bit cumbersome to create, so I think having it as a direct argument would be nice to have.

PR #3493 (2023-09-27): Add `indices` parameter to python Mesh3D api to skip mesh properties

### What

* Fixes #3473

and renamed `vertex_indices` to `indices` because otherwise it looks like a property of the vertex (akin to `vertex_color` etc.)

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3493) (if applicable)

- [PR Build Summary [... cut]
commit 45910dfb (2023-09-27): Add `indices` parameter to python Mesh3D api to skip mesh properties (#3493)

### What

* Fixes #3473

and renamed `vertex_indices` to `indices` because otherwise it looks
like a property of the vertex (akin to `vertex_color` etc.)

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3493) (if
applicable)

- [PR Build [... cut]
closed 2023-09-27 by Wumpf