rerun-io/rerun #3485: Infinitely growing bounding box when changing time

roym899, 2023-09-26:
**Describe the bug**
Here's an rrd file to demonstrate the problem: [gdrive](https://drive.google.com/file/d/1kwOe9YrKpgOmzrm4jU-qkTk94LEA7yyI/view?usp=sharing)
Should work with this prerelease
```bash
pip install --pre --force-reinstall -f https://build.rerun.io/commit/6cf6902/wheels --upgrade rerun-sdk
```

**To Reproduce**
Scroll timeline back and forth and observe the bounding box growing and the view breaking at some point. 
I guess it's related to there being 2D points outside the camera frustum affecting the bounding box, which in turn increases the camera frustums, which in turn increases the spread of these points. But I haven't tested much yet.

**Expected behavior**
Bounding box shouldn't keep growing.

emilk, 2023-10-09:
The correct fix is that the camera frustum should NOT affect the bounding box.

The bounding box should be the bounding box of _data_, not of a visualization like the camera frustum

jleibs, 2023-10-10:
> The correct fix is that the camera frustum should NOT affect the bounding box.

I don't agree with this assessment.  From what I'm seeing, the display of the camera **frustum** is not the main problem. The main problem stems from the projection of 2D data into the 3D scene.  Specifically with 2D points that are interpreted as being in pixel coordinates, but have values like -600 or +800 when the pinhole model indicates a resolution of 256x256.

This is specifically the data `world/input_*/image/secondary_0`.

Additionally for the `world/prediction/image/uv` it looks like something goes extra wrong where we have timeless data for `uv` at a point when we don't have timeless data for the Pinhole and subsequently it gets projected in a weird way.  I think we can probably say 2D data should never get projected into a 3D space if a pinhole isn't defined.

jleibs, 2023-10-10:
I believe the problem with `world/prediction/image/uv` is another variant of: https://github.com/rerun-io/rerun/issues/3092

emilk, 2023-10-10:
I agree - nothing on the 2D projection surface should affect the bounding box

Wumpf, 2024-02-15:
That rrd file no longer works, but the camera projection surface no longer affects the bounding box, only the camera position
https://github.com/rerun-io/rerun/blob/main/crates/re_space_view_spatial/src/visualizers/cameras.rs#L182
So I think this is done!

closed 2024-02-15 by Wumpf