rerun-io/rerun #3502: No helper enum for MediaType in Python

nikolausWest, 2023-09-27:
When using the `rr.TextDocument` archetype to log markdown text in python, it would be useful with a helper enum (in the global namespace) to set the correct mediatype. Rust users have `MediaType::markdown()` but Python users currently need to know to use `mediatype="text/markdown"`. Even writing this issue I had to double check the doc example several times because I forgot what the string was exactly.

teh-cmc, 2023-09-28:
They're there https://github.com/rerun-io/rerun/blob/main/rerun_py/rerun_sdk/rerun/components/media_type_ext.py#L10-L68 :thinking:

nikolausWest, 2023-09-28:
> They're there https://github.com/rerun-io/rerun/blob/main/rerun_py/rerun_sdk/rerun/components/media_type_ext.py#L10-L68 🤔

It doesn't show up in global namespace:
[image]

nikolausWest, 2023-09-28:
Related, I also can't do `rr.components.MediaType`. I think we need to import the `components` module into the global scope as well

closed 2023-09-28 by teh-cmc