rerun-io/rerun #3504: Make sure archetype docs are correct

nikolausWest, 2023-09-27:
After landing
- https://github.com/rerun-io/rerun/issues/2775
- https://github.com/rerun-io/rerun/issues/3437

Take another pass over all archetype pages and make sure:
- Text is correct and descriptive
- We use the correct "pluralness" (Arrow3D vs Arrows3D) everywhere (title, and in text) 
- We link to the correct Python docs (Arrows3D instead of log_arrow)
- Includes at least one code example
- Each code example has a screenshot

Wumpf, 2023-09-29:
* Related to #3540

emilk, 2023-10-03:
Make sure each `Ext` class links back to the original with doclinks, i.e:

```py
class Boxes3DExt:
    """Extension methods for [Boxes3D][rerun.archetypes.Boxes3D]."""
```

emilk, 2023-10-03:
Clean up the CSS for images for the Python API docs

PR #3648 (2023-10-04): Archetype documentation polish 
### What

* Fixes #3504
* Introduce `!api` attribute to make some examples show up only in the api doc
* Reduce api doc example count to 1 on all archetypes
* some example tweaks
* formulation tweaks
* additional cross references to api docs
* tons of images and example title updates

The only archetypes that don't follow strictly the rule of "all examples have an image and everything exactly one example in the api doc" are:
* Text Document is not embedded in any api doc since black chokes on the embedded markdown
* clear recursive doesn't have an image because  [... cut]
commit 2e2e58ae (2023-10-04): Archetype documentation polish  (#3648)

### What

* Fixes #3504
* Introduce `!api` attribute to make some examples show up only in the
api doc
* Reduce api doc example count to 1 on all archetypes
* some example tweaks
* formulation tweaks
* additional cross references to api docs
* tons of images and example title updates

The only archetypes that don't follow strictly the rule of "all examples
have an image and everything exactly one example in the api doc" are:
* Text Document is not embedded in any api doc since black chokes on the
embedded markdown
* clear recursive doesn't have an image [... cut]
closed 2023-10-04 by Wumpf