rerun-io/rerun #3510: TextDocument doesn't handle "visible history"

nikolausWest, 2023-09-28:
It's currently possible to set visible history for `Text` components that are shown in a `TextDocument` view, but nothing changes in the view when you do. A reasonable solution would be to either disable the option and have a separate view type that aggregates multiple `Text` instances, or to just append the strings returned by visual history.

### How to reproduce
```python
import rerun as rr

rr.init("rerun_example_text_stream", spawn=True)

rr.log("text_stream", rr.TextDocument("The start of "))
rr.log("text_stream", rr.TextDocument("a longer "))
rr.log("text_stream", rr.TextDocument("sentence that "))
rr.log("text_stream", rr.TextDocument("ends with."))
```
[image]

abey79, 2023-11-03:
The plan is to add this at the space view level (instead of datagroups/entities like in 2d/3d space views).

abey79, 2023-11-06:
Here: https://github.com/rerun-io/rerun/issues/4044

It was a post-stand-up decision.