rerun-io/rerun #3513: The Python SDK should never raise an exception

nikolausWest, 2023-09-28:
We used to try very hard to make sure the Python SDK never raised an exception to user code but some of this got lost in the move to the new api.

We should:
- Catch all exceptions that happen inside Rerun objects and methods so that using Rerun never causes your program to crash.
- Instead always do the best we can with the data (usually just dropping it) and logging a Warning.
- We need to take special care to handle exceptions in a reasonable way inside constructors of archetypes, components, and datatypes.
    - The right thing will usually be to just construct it as empty and log a warning

teh-cmc, 2023-09-28:
> * We need to take special care to handle exceptions in a reasonable way inside constructors of archetypes, components, and datatypes.    
>   * The right thing will usually be to just construct it as empty and log a warning

Poisoning seems like the most natural approach for this.
E.g. if every archetype constructor returns a `AsComponents`, we can introduce a `PoisonedAsComponents()` object that does the right thing when it gets logged: i.e. don't actually log the data but log a warning instead.

nikolausWest, 2023-09-28:
> Poisoning seems like the most natural approach for this.
> E.g. if every archetype constructor returns a AsComponents, we can introduce a PoisonedAsComponents() object that does the right thing when it gets logged: i.e. don't actually log the data but log a warning instead.

Not sure `__init__ ` methods can return objects of a different type

jleibs, 2023-09-28:
Some additional notes from discussion:
 - This handling will apply to all Archetypes (and maybe all archetype-like things re-exported to the top-level rerun namespace).
 - Components and Datatypes will still raise exceptions so they can be used in python object-centric code (e.g. as part of user-custom archetypes) without needing to globally opt out of strict mode.
 - Where possible we will coerce to a RerunValueError(ValueError).
 - Archetype constructors will also be extended to include a `strict` kwarg to override the default behavior in one-off situations.
 - We *won't* catch the case of using invalid kwargs and let the interpreter raise the standard "unexpected keyword argument" - anything else will make migration and API changes too painful.

jleibs, 2023-09-28:
Any component that produces an exception when logging will result in an empty array of that component being logged instead.  Later we could potentially introduce a way of tracking "bad data" all the way through the data pipeline.

closed 2023-09-30 by jleibs