rerun-io/rerun #3526: `rr.Clear.flat()` not showing in autocomplete

roym899, 2023-09-28:
**Describe the annoyance**
`rr.Clear.flat()` doesn't seem to be picked up by the language server. `rr.Clear.recursive()` does.

**Additional context**
```
rerun_py 0.9.0-alpha.4 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu 29ab0e3, built 2023-09-28T17:19:31Z
```

Wumpf, 2023-10-03:
I get neither picked up, only the `recursive` *field* shows up. Maybe that's what you saw?

The problem is presumably that we add recursive & flat like so:
```
    @staticmethod
    def deferred_patch_class(cls: Any) -> None:
        def flat() -> cls:
            return cls(recursive=False)

        def recursive() -> cls:
            return cls(recursive=True)

        cls.flat = flat
        cls.recursive = recursive
```

roym899, 2023-10-03:
Yeah, you're right, neither shows up as a function.

PR #3618 (2023-10-03): Fix `recursive` python method name-clash & fix autocomplete thereof
### What

* Fixes #3526

`recursive` and `flat` are now regular static methods
had to rename `recursive` to flat because of a nameclash (how did this ever work).

Nice sideeffect is that these methods are now also properly documented

[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tes [... cut]
commit 02a8ad8f (2023-10-03): Fix `recursive` python method name-clash & fix autocomplete thereof (#3618)

### What

* Fixes #3526

`recursive` and `flat` are now regular static methods
had to rename `recursive` to flat because of a nameclash (how did this
ever work).

Nice sideeffect is that these methods are now also properly documented

[image]

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I [... cut]
closed 2023-10-03 by Wumpf