rerun-io/rerun #3527: Viewer hangs upon rearranging tiles

roym899, 2023-09-28:
**Describe the bug**
The viewer hangs when rearranging the tiles and does not recover (see video below). Not sure how easy it will be to reproduce this issue, but at least for this example it seems to happen quite reliably. It happens for me every time I restarted and while it didn't crash immediately after resetting, rearranging the tiles it happened again.

It seems to happen *only* when putting the 3D view into the bottom row of the left-most column.

**Video**

https://github.com/rerun-io/rerun/assets/9785832/5c1bc8fc-47b1-4f9e-82f9-f349807e055d

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
```
rerun_py 0.9.0-alpha.4 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu 29ab0e3, built 2023-09-28T17:19:31Z
```

nikolausWest, 2023-09-28:
Do you get any useful logs if you start the viewer with `RUST_LOG=debug rerun` ?

roym899, 2023-09-28:
Nope, no output at all when it hangs.

nikolausWest, 2023-09-28:
Could you share the rrd file?

emilk, 2023-09-29:
I can reproduce

emilk, 2023-09-29:
Found it

closed 2023-09-29 by Wumpf