rerun-io/rerun #3553: Logging a `SegmentationImage` with an underlying float buffer is a trap

teh-cmc, 2023-09-29:
Consider this:
```py
rr.log("video/mask", rr.SegmentationImage(segmentation_mask.astype(np.uint8)))
```

Without the explicit cast, this numpy tensor is actually a bunch of floats, and so the viewer will fail to interpret the segmentation mask at viewing time.

We should either fail/warn at logging time or do the casting automatically or _something_.

emilk, 2023-10-02:
Let's see if we can do a quick-fix in Rust, checking if they are all integers in `[0, 2^16)`, or else warn.

…or we do it quickly in Python if its too much work for 0.9

jleibs, 2023-10-02:
https://github.com/rerun-io/rerun/pull/3608 will fix this in the short-term.

https://github.com/rerun-io/rerun/issues/3609 is a longer-term fix for after 0.9