rerun-io/rerun #3558: `cargo test -p re_sdk --lib fallbacks` crashes on aarch64 Mac

Wumpf, 2023-09-29:
potentially other platforms as well, but the crash seems to be happen in platform specific code, so assuming this is only apple silicon for now.

Callstack under debugger indicates that it's failing inside a panic during thread cleanup.
```
__abort (@__abort:43)
abort (@__abort:3)
std::sys::unix::abort_internal (@<std::sys::unix::locks::pthread_mutex::AllocatedMutex as std::sys_common::lazy_box::LazyInit>::init:3)
std::panicking::rust_panic_with_hook (@std::panicking::rust_panic_with_hook:127)
std::panicking::begin_panic_handler::{{closure}} (@std::panicking::begin_panic_handler::{{closure}}:54)
std::sys_common::backtrace::__rust_end_short_backtrace (@std::sys_common::backtrace::output_filename:3)
rust_begin_unwind (@<std::panicking::begin_panic_handler::PanicPayload as core::panic::BoxMeUp>::take_box:3)
core::panicking::panic_fmt (@core::panicking::panic_nounwind_fmt:3)
core::result::unwrap_failed (@core::slice::index::slice_start_index_len_fail:3)
core::result::Result<T,E>::expect (@std::panicking::default_hook:195)
std::thread::local::LocalKey<T>::with (@std::panicking::default_hook:186)
std::io::stdio::set_output_capture (@std::panicking::default_hook:186)
std::panicking::default_hook (@std::panicking::default_hook:186)
<alloc::boxed::Box<F,A> as core::ops::function::Fn<Args>>::call (@test::test_main::{{closure}}:139)
test::test_main::{{closure}} (@test::test_main::{{closure}}:136)
<alloc::boxed::Box<F,A> as core::ops::function::Fn<Args>>::call (@std::panicking::rust_panic_with_hook:136)
std::panicking::rust_panic_with_hook (@std::panicking::rust_panic_with_hook:133)
std::panicking::begin_panic_handler::{{closure}} (@std::panicking::begin_panic_handler::{{closure}}:54)
std::sys_common::backtrace::__rust_end_short_backtrace (@std::sys_common::backtrace::output_filename:3)
rust_begin_unwind (@<std::panicking::begin_panic_handler::PanicPayload as core::panic::BoxMeUp>::take_box:3)
core::panicking::panic_fmt (@core::panicking::panic_nounwind_fmt:3)
core::panicking::panic_display (@core::panicking::panic_fmt:3)
core::panicking::panic_str (@core::option::expect_failed:7)
core::option::expect_failed (@core::option::expect_failed:7)
core::option::Option<T>::expect (@std::thread::park:38)
std::thread::current (@std::thread::park:32)
std::thread::park (@std::thread::park:32)
crossbeam_channel::context::Context::wait_until (/Users/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/crossbeam-channel-0.5.8/src/context.rs:177)
crossbeam_channel::flavors::zero::Channel<T>::recv::{{closure}} (/Users/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/crossbeam-channel-0.5.8/src/flavors/zero.rs:323)
crossbeam_channel::context::Context::with::{{closure}} (/Users/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/crossbeam-channel-0.5.8/src/context.rs:52)
crossbeam_channel::context::Context::with::{{closure}} (/Users/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/crossbeam-channel-0.5.8/src/context.rs:60)
std::thread::local::LocalKey<T>::try_with (@crossbeam_channel::flavors::zero::Channel<T>::recv:115)
crossbeam_channel::context::Context::with (/Users/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/crossbeam-channel-0.5.8/src/context.rs:55)
crossbeam_channel::flavors::zero::Channel<T>::recv (/Users/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/crossbeam-channel-0.5.8/src/flavors/zero.rs:310)
crossbeam_channel::channel::Receiver<T>::recv (/Users/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/crossbeam-channel-0.5.8/src/channel.rs:816)
re_log_types::data_table_batcher::DataTableBatcherInner::flush_blocking (/Users/andreas/dev/rerun-io/rerun/crates/re_log_types/src/data_table_batcher.rs:356)
re_log_types::data_table_batcher::DataTableBatcher::flush_blocking (/Users/andreas/dev/rerun-io/rerun/crates/re_log_types/src/data_table_batcher.rs:326)
<re_sdk::recording_stream::RecordingStreamInner as core::ops::drop::Drop>::drop (/Users/andreas/dev/rerun-io/rerun/crates/re_sdk/src/recording_stream.rs:461)
core::ptr::drop_in_place<re_sdk::recording_stream::RecordingStreamInner> (@core::ptr::drop_in_place<re_sdk::recording_stream::RecordingStreamInner>:13)
core::ptr::drop_in_place<core::option::Option<re_sdk::recording_stream::RecordingStreamInner>> (@alloc::sync::Arc<T>::drop_slow:12)
alloc::sync::Arc<T>::drop_slow (@alloc::sync::Arc<T>::drop_slow:7)
<alloc::sync::Arc<T> as core::ops::drop::Drop>::drop (@std::sys::common::thread_local::fast_local::destroy_value:25)
core::ptr::drop_in_place<alloc::sync::Arc<core::option::Option<re_sdk::recording_stream::RecordingStreamInner>>> (@std::sys::common::thread_local::fast_local::destroy_value:23)
core::ptr::drop_in_place<re_sdk::recording_stream::RecordingStream> (@std::sys::common::thread_local::fast_local::destroy_value:23)
core::ptr::drop_in_place<core::option::Option<re_sdk::recording_stream::RecordingStream>> (@std::sys::common::thread_local::fast_local::destroy_value:23)
core::ptr::drop_in_place<core::cell::UnsafeCell<core::option::Option<re_sdk::recording_stream::RecordingStream>>> (@std::sys::common::thread_local::fast_local::destroy_value:23)
core::ptr::drop_in_place<core::cell::RefCell<core::option::Option<re_sdk::recording_stream::RecordingStream>>> (@std::sys::common::thread_local::fast_local::destroy_value:23)
core::ptr::drop_in_place<core::option::Option<core::cell::RefCell<core::option::Option<re_sdk::recording_stream::RecordingStream>>>> (@std::sys::common::thread_local::fast_local::destroy_value:15)
core::mem::drop (@std::sys::common::thread_local::fast_local::destroy_value:13)
std::sys::common::thread_local::fast_local::destroy_value::{{closure}} (@std::sys::common::thread_local::fast_local::destroy_value:13)
core::ops::function::FnOnce::call_once (@std::sys::common::thread_local::fast_local::destroy_value:6)
<core::panic::unwind_safe::AssertUnwindSafe<F> as core::ops::function::FnOnce<()>>::call_once (@std::sys::common::thread_local::fast_local::destroy_value:6)
std::panicking::try::do_call (@std::sys::common::thread_local::fast_local::destroy_value:6)
std::panicking::try (@std::sys::common::thread_local::fast_local::destroy_value:6)
std::panic::catch_unwind (@std::sys::common::thread_local::fast_local::destroy_value:6)
std::sys::common::thread_local::fast_local::destroy_value (@std::sys::common::thread_local::fast_local::destroy_value:6)
std::sys::unix::thread_local_dtor::register_dtor::run_dtors (@std::sys::unix::thread_local_dtor::register_dtor::run_dtors:33)
invocation function for block in dyld4::RuntimeState::_finalizeListTLV(void*) (@invocation function for block in dyld4::RuntimeState::_finalizeListTLV(void*):17)
dyld4::RuntimeState::_finalizeListTLV(void*) (@dyld4::RuntimeState::_finalizeListTLV(void*):32)
_pthread_tsd_cleanup (@_pthread_tsd_cleanup:125)
```

Wumpf, 2023-09-29:
@jparismorgan that's the error you ran into as well over on #3530. What platform were you running on?

jparismorgan, 2023-09-29:
> @jparismorgan that's the error you ran into as well over on #3530. What platform were you running on?

I'm on an Apple M2 Pro running MacOS 13.3 (22E252).

closed 2023-10-20 by emilk