rerun-io/rerun #3573: Don't show indicators on hover

nikolausWest, 2023-10-01:
Hovering over data in a space view can easily become quite cluttered (see example). We should therefore hide indicator components as they aren't really the data you are interested in inspecting in a view.

[image]

nikolausWest, 2023-10-01:
I also wonder about "hover position" for 3D points when hover in 2D. I think we should remove it but I may be wrong. I definitely don't understand it.

Wumpf, 2023-10-02:
Afaik showing the position was a user request:
* #3226
Also not happy with what it looks like though in the ui. There's some more notes on the topic here
* #2843

nikolausWest, 2023-10-02:
> Also not happy with what it looks like though in the ui. There's some more notes on the topic here

In my opinion the current implementation makes things worse more than better. It's not worth the added clutter when it's hard to understand

closed 2023-10-02 by emilk