rerun-io/rerun #3584: Use `camera_xyz` consistently in examples where we log pinhole + view coordinates

jleibs, 2023-10-02:


jleibs, 2023-10-02:
There are many places where this is *almost* what we are doing.

A common pattern looks like:
```
        rr.log("camera", rr.ViewCoordinates.RDF)
        # Log camera intrinsics
        assert camera.model == "PINHOLE"
        rr.log(
            "camera/image",
            rr.Pinhole(
                resolution=[camera.width, camera.height],
                focal_length=camera.params[:2],
                principal_point=camera.params[2:],
            ),
        )
```

Moving the pinhole declaration down into camera/image means if you create a space-view located at the camera root you get the wrong default view coordinates.

emilk, 2023-10-03:
What is even the effect of `rr.log("camera", rr.ViewCoordinates.RDF)` here? I don't think it does anything

emilk, 2023-10-03:
All examples uise `camera_xyz` where appropiate

closed 2023-10-03 by emilk