rerun-io/rerun #3606: UI handling for "user.components." prefix

jleibs, 2023-10-02:
When deprecating `ext=` and introducing `rr.AnyValues` we started using the prefix "user.components.<name>"

This is a little bit on the verbose side. We should either pick a different prefix or special-case it in the UI.  Would be nice to do something like highlight it in a different color, etc.

nikolausWest, 2023-10-02:
I think we should go with something much shorter, like for instance `any.`

emilk, 2023-10-04:
It is currently `any.values.MyComponent`

emilk, 2023-10-04:
`AnyBatchValue` docs still says:

> The component will be named "user.components.<NAME>".

emilk, 2023-10-04:
Alternative naming suggestions for `AnyValues`:
* `Custom`
* `Dynamic`

jleibs, 2023-10-04:
Decided to simply drop the prefix for now.

https://github.com/rerun-io/rerun/issues/3661 may force us to change the way this works in the future.

closed 2023-10-04 by jleibs