rerun-io/rerun #3612: Markdown heading levels all have the same font size

roym899, 2023-10-03:
See title. I'd expect smaller font size for higher heading levels.

**Additional context**
```
rerun_py 0.9.0-alpha.6 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu bde24df, built 2023-10-02T12:51:19Z
```

haritha208, 2023-10-03:
assign the issue to me,currently i am working on ui & ux

nikolausWest, 2023-10-09:
@haritha208: Do you know how to fix it? I believe it requires a fix in egui and/or egui_commonmark

abey79, 2023-10-11:
repro:

```python
import rerun as rr

rr.init("md", spawn=True)

rr.log(
    "my_points",
    rr.TextDocument("""
# Heading level 1

## Heading level 2

### Heading level 3

#### Heading level 4

##### Heading level 5

###### Heading level 6

Body text
 """, media_type=rr.MediaType.MARKDOWN)
)
```

result:

[image]

I believe this might be due to some interaction between the way `egui_commonmark` sets up styling and our own `re_ui` customisations:
[image]

abey79, 2023-11-07:
Fixed by https://github.com/lampsitter/egui_commonmark/pull/20

PR #4178 (2023-11-08): Fix markdown heading size
### What

- Take advantage of https://github.com/lampsitter/egui_commonmark/pull/20 to use "correct" headings in in-app quick start guides.
- Bump egui_commonmark to 0.9.2 which includes the PR above
- Fixes #3612

[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/4178) (if applica [... cut]
commit a586ae4d (2023-11-08): Fix markdown heading size (#4178)

### What

- Take advantage of
https://github.com/lampsitter/egui_commonmark/pull/20 to use "correct"
headings in in-app quick start guides.
- Bump egui_commonmark to 0.9.2 which includes the PR above
- Fixes #3612

<img width="1305" alt="image"
src="https://github.com/rerun-io/rerun/assets/49431240/a4befe22-2675-4a8f-a5ca-38cec3141b13">

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I'v [... cut]
closed 2023-11-08 by abey79