rerun-io/rerun #3643: LineStrips2D and LineStrips3D doesn't correctly handle a single strip

jleibs, 2023-10-03:
Repro: remove the extra `[]` from around points in the code example:
```
points = [
    [0, 0, 0],
    [0, 0, 1],
    [1, 0, 0],
    [1, 0, 1],
    [1, 1, 0],
    [1, 1, 1],
    [0, 1, 0],
    [0, 1, 1],
]

rr.log("strip", rr.LineStrips3D(points))
```

It looks like we interpret this as 8 line strips of length 1 instead of 1 line strip of length 8.

SeaOtocinclus, 2023-10-03:
I think from a user perspective it summarizes the convenience of usage as supporting use case as:
- 1 single list of points
- or multiple lists of points

So update the doc also to showcase those

https://ref.rerun.io/docs/python/prerelease/package/rerun/__init__/#rerun.LineStrips3D

It would ending up:

- Simple usage (1 group - no need to use [[]])
```
points =  [0, 0, 0],  [0, 0, 1]

rr.log("strip", rr.LineStrips3D([points]))
```

- Many individuals line strips (4 groups)
```
points = np.array(
            [
                [[0, 0, 0], [0, 0, 1]],
                [[1, 0, 0], [1, 0, 1]],
                [[1, 1, 0], [1, 1, 1]],
                [[0, 1, 0], [0, 1, 1]],
            ],
        )
rr.log("strip", rr.LineStrips3D([points]))
```

SeaOtocinclus, 2023-10-03:
Also does such a behavior propagates to other `components`?

It seems so, since I can read `rr.log("points", rr.Points3D([[0, 0, 0], [1, 1, 1]]))` on the page https://ref.rerun.io/docs/python/prerelease/package/rerun/__init__/#rerun.Points3D but we also would like to use
rr.log("points", rr.Points3D([0, 0, 0], [1, 1, 1]))`

Wumpf, 2023-10-04:
the snippet @jleibs posted we can fix easily enough but making `rr.log("points", rr.Points3D([0, 0, 0], [1, 1, 1]))` is much more challenging to get to work unambiguously since the first parameter of Points3D is  "`Position3DArrayLike`", i.e. something that we can interpret as 3D positions

commit f96fe2d8 (2023-10-04): Handle single line strips better (#3668)

### What

* Fixes #3643 

And disallow line strips from flat number lists. I though this would
break legacy API, but according to legacy api documentation it
shouldn't.

Tried to make the linestrip method simpler but thoroughly failed in that
quest. Naturally, added plenty more unit tests to see if it does what
it's supposed to.

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] ~I'v [... cut]
closed 2023-10-04 by jleibs