rerun-io/rerun #3645: Transform3D should have an `affine3d=` helper.

jleibs, 2023-10-04:
A 3x4 affine matrix is a common representation. Would be nice to be able to pass this directly to `rr.Transform3D` instead of needing to:
```
rr.Transform3D(translation=transform[:,3], mat3x3=transform[:,0:3])
```

Wumpf, 2023-10-05:
* Related to..
   * #3559
   * #2868

emilk, 2023-10-09:
If we call the parameter `affine3d` I think it should accept both a 3x4 and a 4x4 matrix, and in the latter case just ignore the last row

(or is it 4x3? or is it ignore the last column?)

Wumpf, 2023-10-24:
I believe we're now in a good spot to just support 3x4 and 4x4 directly as new union variants. I'd prefer we solve it like that for all languages instead of doing a python specific thing (beyond special named parameters)

Wumpf, 2024-07-31:
closing this as won't do / duplicate:
we do have to add a matrix 4x4 and matrix 3x4, but the way we're modelling data right now (and potentially in the future) a `affine3d=` helper doesn't make sense because it implies introducing an `Affine3D` 

For additional matrices see:
* https://github.com/rerun-io/rerun/issues/3559

closed 2024-07-31 by Wumpf