rerun-io/rerun #3647: Improve Python API docs version selection menu

emilk, 2023-10-04:
If you go to https://ref.rerun.io/docs/python/ you can select a version from a drop-down.

Currently at the top is `0.8.2`. At the bottom you'll find:

[image]

What is `test`?

What is `prerelease`? Is this the weekly alpha?

Is `HEAD` an idiomatic name in Python documentation? Can we call it `nightly` instead?

I would like it if we had `0.9.0-alpha.6` in that list. Not _all_ alphas, just the _latest_ alpha, spelled out explicitly. Or at least that if you select `nightly` or whatever, that it spells out the version for it.

jprochazk, 2023-10-04:
Note that [`latest`](https://ref.rerun.io/docs/python/latest) exists, just that link doesn't work because `latest` points to `0.8.2` right now.

abey79, 2023-10-04:
> Is HEAD an idiomatic name in Python documentation? Can we call it nightly instead?

Most python docs are hosted on readthedocs.io, and the idiomatic name there is `latest` = `main`, and the version selector usually defaults to the latest released version. There usually isn't a handle for the "latest release".

jprochazk, 2023-10-04:
> What is test?

That's from when someone manually dispatched the docs release job, can safely be deleted

> What is prerelease? Is this the weekly alpha?

That's the latest alpha, like the one we just released. I'll change it to be called just `latest alpha`. I guess there's no need for that if we have the latest alpha version on that list.

emilk, 2023-10-04:
Let's go with

* `nightly` - nightly, or every commit to main
* `stable` - actual "real" release
* `dev` - latest `alpha` or `rc`

jprochazk, 2023-10-05:
This is done, just missing final cleanup of the old versions in the dropdown. We'll do that after the 0.9 release.

closed 2023-10-09 by emilk