rerun-io/rerun #3661: Formalize the meaning of fully-qualified type names

jleibs, 2023-10-04:
We currently follow a convention of adding name prefixes for:
 - `rerun.archetypes'
 - `rerun.components`
 - `rerun.datatypes`
 
 As of https://github.com/rerun-io/rerun/pull/3660, we no longer add any prefixes to user components.
 
The `rerun.components` prefixes, specifically, make it all the way through the data model and would otherwise show up as the component/field names in the UI. As such we have special handling to strip off the `rerun.components` and to annotate components ending with `Indicator` differently.

If these conventions are going to carry special meaning we should capture that with some form of RFC.
 - What makes a valid name? Are there reserved/illegal characters?
 - Do we always want to require an "org" prefix such as rerun?
 - These names seem likely to play a role in plugin-handling / dispatch in the future.
 - Will there be special handling of the intermediate type organization? Is there a good reason for someone to match, e.g. "*.components.*".

teh-cmc, 2023-10-10:
I really don't like the idea of having semantics attached to fully-qualified names.

It obviously is the case today, but to me that's always been more of a crutch than anything, so we can keep the ball rolling until we can get an actual runtime schema registry with proper schema metadata / extensions / etc running:
- #1809 

Once we have this registry (and we're close: #3741), introspecting the schema associated with a fully-qualified name becomes `O(1)` and queries like:
```
registry.get_schema("rerun.components.Point3D")?.is_builtin()
registry.get_schema("rerun.components.Point3D")?.is_component()
registry.get_schema("rerun.components.Point3D")?.is_indicator()
```
become trivial.

Similarly, I would expect we won't be looking for indicator components by parsing their suffixes in the future: there should be a `RERUN:indicator` marker in their schema metadata or something.

I do think it's perfectly fine to make use of known prefixes for trivial things though (e.g. UI hints), even in a world with a runtime registry, but that's as far as they should go semantic-wise.

Plugin handling and dispatching, as you say, is where I expect fqnames to really play an important role.
Especially once we start opening plugins to everyone, at which point namespacing conventions will become super important.

teh-cmc, 2023-10-11:
- Related: https://github.com/rerun-io/rerun/pull/3790/files#r1354214405
> [image]