rerun-io/rerun #3677: Bad heuristic results in adding inappropriate segmentation to 3D scenes

jleibs, 2023-10-04:
Repro:
```
"""Log a pinhole and a random image."""
import numpy as np
import rerun as rr

rr.init("rerun_example_pinhole", spawn=True)
rng = np.random.default_rng(12345)

image = rng.uniform(0, 255, size=[3, 3, 3])
rr.log("world/camera/image", rr.Pinhole(focal_length=3, width=3, height=3))
rr.log("world/camera/image", rr.Image(image))

image = rng.uniform(0, 255, size=[3, 3, 1])
rr.log("seg", rr.SegmentationImage(image))
```

[image]

Note this *only* happens with `rr.SegmentationImage` but not with `rr.Image` so there's probably some place we need to do a more complete check.

emilk, 2023-10-04:
so `seg` and `world/camera/image` are added to the same spaceview, despite them not having a common ancestor O.o that's really strange

Wumpf, 2023-10-05:
that part isn't _that_ strange since as of 0.8 (or was it 0.7 already?) we sometimes spawn a single space view at the root

PR #3681 (2023-10-05): Fix presence of segmentation images and tensor causing generation of root space view
### What

* Fixes #3677

Definitely need to come up with something better to steer the heuristics - we made good strides in space view refactor and recent advances in System heuristics, but there's still a lot to be desired!

In the test scene there's still the issue of a unexplicably large accumulated bounding box which causes the camera to zoom out a lot. This is most likely caused by an initially bad heuristic for the frustum length. I suspect this is a problem we already had in 0.8 and might be fairly hard [... cut]
commit b26ca172 (2023-10-05): Fix presence of segmentation images and tensor causing generation of root space view (#3681)

### What

* Fixes #3677

Definitely need to come up with something better to steer the heuristics
- we made good strides in space view refactor and recent advances in
System heuristics, but there's still a lot to be desired!

In the test scene there's still the issue of a unexplicably large
accumulated bounding box which causes the camera to zoom out a lot. This
is most likely caused by an initially bad heuristic for the frustum
length. I suspect this is a problem we already had in 0.8 and might be
fa [... cut]
closed 2023-10-05 by emilk